Ansem-SoD/Picofly #5: Last update 2.72 does not update

curcius, 2023-05-18:
I can't do the update by picofly payload it says it loaded the update.bin but when I restart the switch it restarts in ofw and the chip is flashing white, I restart and it gives the glich but it continues in the previous version in case 2.67. How should I do to update? only by usb?

Ansem-SoD, 2023-05-19:
This might be silly but did you use the latest update.bin?

drty2k, 2023-05-19:
i have same problem, try many times - after power off it booting ofw, after next reboot glitch work fine but firmware stay same = 2.67

curcius, 2023-05-19:
> i have same problem, try many times - after power off it booting ofw, after next reboot glitch work fine but firmware stay same = 2.67

Yes I downloaded the latest update.bin

drty2k, 2023-05-19:
get update.bin from this place and all work fine)
https://gbatemp.net/threads/picofly-a-hwfly-switch-modchip.622701/page-78#post-10090767

mistervampi, 2023-05-19:
Same here!! Even with GBATemp BIN
V2 with Kioxia (Toshiba) NAND

curcius, 2023-05-19:
> get update.bin from this place and all work fine) https://gbatemp.net/threads/picofly-a-hwfly-switch-modchip.622701/page-78#post-10090767

I did the gbatemp procedure and it worked but I did two more steps to make it work 

-show firmware information
- update firmware from SD card (place update.bin into the root folder)
- rollback to the backup firmware slot
- reset learning statistics
- dump / write sdloader
- back to hekate
- shutdown the consol for the hekate
- and 2.72 updated 

😃

curcius, 2023-05-19:
> get update.bin from this place and all work fine) https://gbatemp.net/threads/picofly-a-hwfly-switch-modchip.622701/page-78#post-10090767

Thank you so much

curcius, 2023-05-19:
Sorted out

curcius, 2023-05-19:
Solved

closed 2023-05-19 by curcius

ad1095445, 2025-01-19:
Bro,i am in 2.65,and i want upate to 2.8. it's always faild ,and firmware info show "Fuse count 0".what should i do.