Ansem-SoD/Picofly #7: 2.72 grb to rgb doesn’t work

Gabry00101, 2023-05-19:
In the 2.72 after the upload of the firmware appear always the green light, also with grounded pin

dicamarques14, 2023-05-19:
Since 2.7 the led doesnt need that bridge, now the debug is by long and short yellow pulses

https://gbatemp.net/threads/picofly-a-hwfly-switch-modchip.622701/post-10090767

= is long pulse, * is short pulse:

= USB flashing done

** RST is not connected
*= CMD is not connected
=* D0 is not connected
== CLK is not connected

=** eMMC init failure during glitch process
=*= CPU never reach BCT check, should not happen
==* CPU always reach BCT check (no glitch reaction, check mosfet)
=== Glitch attempt limit reached, cannot glitch

=*** eMMC init failure
=**= eMMC write failure - comparison failed
=*=* eMMC write failure - write failed
=*== eMMC test failure - read failed
==** eMMC read failed during firmware update
==*= BCT copy failed - write failure
===* BCT copy failed - comparison failure
==== BCT copy failed - read failure

Gabry00101, 2023-05-19:
Thanks, I've missed that information 😁

Ansem-SoD, 2023-05-21:
updated to 2.73 error codes are in changelog.md

closed 2023-05-21 by Ansem-SoD