Ansem-SoD/Picofly #11: Long short short short

quasicmatt, 2023-05-27:
I’m getting the error code long short short short and while I know what the error is thanks to the guide I don’t know how it’s supposed to be fixed

Ansem-SoD, 2023-05-28:
what switch? 

If oled check your dat0 and clk points

quasicmatt, 2023-05-28:
> what switch?
> 
> If oled check your dat0 and clk points

It is a plain switch but I will still check those points in case

Ansem-SoD, 2023-05-28:
curious 🤔 

Wonder if you need the additional SDL SCA points

quasicmatt, 2023-05-29:
Where would the sdl and sca points go on the board cause all the pictures of them I have seen I don’t have on my board

Ansem-SoD, 2023-05-30:
> Where would the sdl and sca points go on the board cause all the pictures of them I have seen I don’t have on my board

The additional points are included in the latest [AIO Guide](https://gbatemp.net/download/a-definitive-picofly-install-guide.37968/download)

closed 2023-06-03 by Ansem-SoD

quasicmatt, 2023-06-06:
Sorry for finally had time to work on it again sdl and sca points didn’t fix issue but also the points it shows in the guide weren’t exactly there on my board so I just solder where it shows even though it looked slightly different