Ansem-SoD/Picofly #14: What is contained in mariko_bct_data

songzhishuo, 2023-06-05:
What is contained in mariko_bct_data, is this part planned to be open source?

My understanding is that the boot boot and the part that drives the screen to display the logo are stored in bct

songzhishuo, 2023-06-05:
I don't quite understand what you mean.

songzhishuo, 2023-06-05:
If I want to change the logo(PicoFly NO SD CARD), where do I need to modify it?

rehius, 2023-06-06:
https://switchbrew.org/wiki/BCT

logo is within payload.h, there is no option to modify the logo for now.
I do not have the source code, its author is against opening the source code.

songzhishuo, 2023-06-06:
> https://switchbrew.org/wiki/BCT
> 
> logo is within payload.h, there is no option to modify the logo for now. I do not have the source code, its author is against opening the source code.

I see the following in the code (write_data(0x1F80, payload, sizeof(payload)); ），But I don't know what payload specifically 。I just wanted to change the logo of my switch when it started up to make it look coolermeans

songzhishuo, 2023-06-06:
who is this proj author ,not you? @rehius

songzhishuo, 2023-06-07:
> There will be an update (hopefully soon) that will change that.

It would be really nice if there was an update that would allow users more freedom. At present, it seems that I cannot modify the payload file, because I don't know how it was compiled.
Will the new version provide the ability to modify the logo?
Will there be any other new function in features?