Ansem-SoD/Picofly #25: Fw 2.73 Sometimes boots to ofw giving different errors each time

Christianacqua, 2023-08-01:
I have the chip since may, everything worked and now works normally but sometimes the chip acts weird, it boots to ofw giving some errors like 1 week ago i was getting emmc init failiure and read failiure, yesterday it did dat0 not connected but after some restarts it started working again, i fixed everything with some restarts. I don’t know why and i would like to fix this permanently and know what the cause is. Thanks

rursache, 2023-08-05:
@Christianacqua had the same issue but i fixed it using the picofly toolbox and selecting reset training data. now it glitches everytime without issues. i didn't opened the console up or did anything else.

Jeqvl0619, 2023-08-05:
Just install the two i2c wires: sda & scl... Something similar happened to me in an OLED, booted into CFW and when power turn OFF and ON again, booted into OFW, and needed to try multiple times to boot into CFW again, I solder the two points and problem solved.

rursache, 2023-08-05:
i haven't installed the modchip myself as i do not have the required soldering skills or tools. i have pictures of the switch opened and the chip installed and upon checking, indeed the i2c wires are not installed. oh well. @Jeqvl0619 any other tips of things I can do?

Jeqvl0619, 2023-08-05:
@rursache no my friend, just that...

Christianacqua, 2023-08-06:
> @Christianacqua had the same issue but i fixed it using the picofly toolbox and selecting reset training data. now it glitches everytime without issues. i didn't opened the console up or did anything else.

Ye i did it too and now its been 5 days and i have no issues at all. Thanks mate

remlei, 2023-10-21:
I also had the same issue as well. The only thing that did fix it for me is basically reflashing the uf2 file on rp2040 and that basically fixed this weird issue that randomly boot to OFW giving you different error codes on 2.70+ firmware when it fails the glitch. Resetting the calibration didnt help at all.

I suspect some hidden bug present with the calibration that keeps it messing up that even resetting the calibration didnt help at all and only reflashing the uf2 file fixed it.

Quas7, 2023-10-21:
I think, that there is no verify function with uf2 flashing. Therefore, bad USB connection or random hickups result also in bad programming.

hongtm77, 2023-11-18:
> install the two i2c wires: sda & scl

hi, I got the same problem. previously tried multiple reboot still able to to go back to cfw, but now almost all the time booting into ofw... might need to try the on my oled, any diagram for this soldering for my oled?

Christianacqua, 2023-11-18:
i would like to give you an update, some days ago it did the bug again on fww 2.75. It lasted 16 hrs. eevery hour i rebooted/power on/off and eventually it worked again. Today i reflashed the fw again and atm no problems so far, i don't know if it's a bug or hardware problem but now it works

hongtm77, 2023-11-18:
> i would like to give you an update, some days ago it did the bug again on fww 2.75. It lasted 16 hrs. eevery hour i rebooted/power on/off and eventually it worked again. Today i reflashed the fw again and atm no problems so far, i don't know if it's a bug or hardware problem but now it works

Hi, so you mean fww 2.75 is better than 2.73? I just flashed 2.73 from 2.74. And its still the same

hongtm77, 2023-11-20:
> Just install the two i2c wires: sda & scl... Something similar happened to me in an OLED, booted into CFW and when power turn OFF and ON again, booted into OFW, and needed to try multiple times to boot into CFW again, I solder the two points and problem solved. 

Hi, how to solder ya? You mean just bridging the 2 points together?

hongtm77, 2023-11-20:
> i would like to give you an update, some days ago it did the bug again on fww 2.75. It lasted 16 hrs. eevery hour i rebooted/power on/off and eventually it worked again. Today i reflashed the fw again and atm no problems so far, i don't know if it's a bug or hardware problem but now it works

Have you tried to reboot multiple times after a reflash? Because i will be constantly rebooting to ubuntu or android so not sure it will be stable or not?

Christianacqua, 2023-11-20:
> Hi, so you mean fww 2.75 is better than 2.73? I just flashed 2.73 from 2.74. And its still the same

they're basically the same but do not install 2.74 as it is buggy iirc, rn after reupdating to 2.75 with the toolbox i'm not facing any problems so far, my conclusion is that rp2040 is a 3$ chip and it is a miracle that we are modding the switch with it so of course it has some bugs, if your diode values are right then it is a bug, try to power off and on (not reboot it did never work for me) till it boots to hekate then as i said re update with 2.75.

hongtm77, 2023-11-21:
I have finally fixed the problem. Previously boot cfw was not consistent, sometimes boot ofw and the error was =* (long short) D0 not connected. And the dat0 cable was not fully connected, i resoldered it and now consistently boot into ofw even restart or turn off and on the switch

Christianacqua, 2023-11-21:
> I have finally fixed the problem. Previously boot cfw was not consistent, sometimes boot ofw and the error was =* (long short) D0 not connected. And the dat0 cable was not fully connected, i resoldered it and now consistently boot into ofw even restart or turn off and on the switch

Ye so the dat0 was a bit loose but not entirely so sometimes it glitched and sometimes no because the connection wasn't strong