Ansem-SoD/Picofly #26: Modo bypass

leocsum, 2023-09-18:
Como tirar do modo ByPass?

Ixolaro, 2023-09-18:
la modalità bypass è una sola volta, se continui ad eseguire l'avvio in OFW, controlla il LED del chip

Ixolaro, 2023-09-18:
Elenco dei codici di errore (= è un impulso lungo, * è un impulso breve):

= Flashing USB terminato

** RST non è connesso

*= CMD non è connesso

=* D0 non è connesso

== CLK non è connesso

=** Errore di inizializzazione eMMC durante il processo di glitch

=*= La CPU non raggiunge mai il controllo BCT, non dovrebbe accadere

==* La CPU raggiunge sempre il controllo BCT (nessuna reazione glitch, controlla il mosfet)

=== Limite tentativi di glitch raggiunto, impossibile glitch

=*** Errore di inizializzazione eMMC

=**= Errore di scrittura eMMC: confronto non riuscito

== Errore di scrittura eMMC: scrittura non riuscita

=*== Errore test eMMC: lettura non riuscita

==** Lettura eMMC non riuscita durante l'aggiornamento del firmware

==*= Copia BCT fallita - scrittura fallita

===* Copia BCT non riuscita - confronto non riuscito

==== Copia BCT non riuscita: lettura non riuscita

Ixolaro, 2023-10-04:
L'avvio di OFW è temporaneo. Sembra che il tuo chip abbia un problema