Ansem-SoD/Picofly #31: error code *== (short long long)

hongtm77, 2023-10-21:
Hi guys, i encountered an error that is not listed in the error code which is short, long, long (*==) any idea what is this mean?

Quas7, 2023-10-21:
from the guide: [PicoFlyGuideV.6.2.pdf](https://github.com/Ansem-SoD/Picofly/blob/cc1099d6e9f659425b132fc73b9153fe07d605f8/PicoFlyGuideV.6.2.pdf)

```
** RST is not connected
*= CMD is not connected
=* D0 is not connected
== CLK is not connected
*** No eMMC CMD1 responce (bad eMMC?)
**= No eMMC block 1 read (should not happen)
*=* No eMMC block 0 read (eMMC init failure?)
*== No eMMC CMD1 request (poor wiring, or dead CPU)
=** eMMC init failure during glitch process
=*= CPU never reach BCT check, should not happen
==* CPU always reach BCT check (no glitch reaction, check
mosfet)
=== Glitch attempt limit reached, cannot glitch
=*** eMMC init failure
=**= eMMC write failure - comparison failed
=*=* eMMC write failure - write failed
=*== eMMC test failure - read failed
==** eMMC read failed during firmware update
==*= BCT copy failed - write failure
===* BCT copy failed - comparison failure
==== BCT copy failed - read failure
```

shakangel, 2024-01-03:
Check the CLK cable. This error happened to me in an OLED where I had inavertandly compressed the CLK line with the APU heatsink. I guess the CLK line was shorting to ground or sending strange signals. Worth the check.

lucasromeiro, 2024-02-01:
Sometimes I have this type of problem and I don't understand how to solve it!
The resistor solder at point A is perfect.... what else could cause this??

kastras, 2024-02-02:
I had the same issue and the problem was the dat0 point, i had the flex touching two points. My switch can´t turn on with chip or without it. If you had the same issue, try removing the flex in dat0 and turning on the switch without chip, if it works its that, and you need to put again the flex in the correct possition.

FabioKastro, 2024-03-24:
This is now happening to me, spent all day yesterday trying to figure out what was wrong, my Switch was working perfectly. Now that i was able to replicate the problem i found out that if you remove the SD Card while the emunand is booted the console will "crash??" and won't turn ON or show anything on the screen. If you force reboot by holding the power button the modchip will throw this error *== No eMMC CMD1 request (poor wiring, or dead CPU) and nothing on the screen, the switch appears to be completely dead. After resoldering everything step by step what seemed to work was resolder the flex on the CPU, the console was no longer dead and turned ON. 

BUT!!! Big BUT!!! 
Like i said, i was able to replicate this problem. Did the exact same, removed the SD card while booted into emunand and Switch no longer turns ON. So im pretty sure this was never a problem with soldering. The battery is now uplugged while i wait for everything to drain and hopefully fix the problem once again.

TheCyberQuake, 2024-06-26:
Based on the finding in another open issue and my own experience, seems like the undocumented *== is usually tied to CLK due to improper handling of the error codes, where CLK isn't actually implemented. 
For OLED, the most common issue causing it seems to be when the user exposes the CLK point by scrapping the solder mask, they accidentally expose part of the surrounding ground plane. Pulling up the ribbon/cable, applying UV solder mask to any exposed ground plane surrounding it, and then soldering the ribbon back down seems to fix it in most cases.

slewantoski, 2024-07-06:
I've had the exact same error code just now.
First could not find it. After some trying i realised that it only happens if i have the APU Shielding / Cooling attached.
Then found this post.

It was my CLK Line connected to the Shielding. Removed some of the tabs on this side and works perfectly now.

TheCyberQuake, 2024-07-06:
> I've had the exact same error code just now.
> First could not find it. After some trying i realised that it only happens if i have the APU Shielding / Cooling attached.
> Then found this post.
>
> It was [...]

Seems like we can confirm it's most often to show if CLK is shorted to ground then

BagraNxt, 2025-01-09:
Found the solution if you encounter such problem. Just replace the modchip or buy an new one. Coz the problem is in the modchip.

slewantoski, 2025-01-09:
Nope, my modchip still works without replacing the chip. I just removed a part from the APU shielding to remove the short to ground from CLK.

TheCyberQuake, 2025-01-09:
> Found the solution if you encounter such problem. Just replace the modchip or buy an new one. Coz the problem is in the modchip.

Nope. Had several of us have already found that error code just means CLK is shorted to ground. 
Your replace fix was likely just the fact that you undid the old one with a bad solder and redid the new one, when you likely could have been fixed by just redoing your solder

baalbeker, 2025-05-28:
Same issue happened to me after creating emunand partition that failed mid install.Wont boot.
Short long long yellow.
Little fed up at this point.

Doesnt work with another modchip,i had one spare.
Prolly will resolder it for the 3rd time poor me

TheCyberQuake, 2025-05-28:
> It wont boot even after removing the chip so i assume cpu is dead

Sounds like you need to go back and make sure you didn't short anything.
Depending on which system, there are plenty of places that you wouldn't be able to tell it's shorted without microscope, and then with it can be difficult.
And if you did short it out, that would be due to the install

baalbeker, 2025-05-29:
Checked with multimeter and it looks like there is short on both ceramic capacitors(sp1/2).
Will replacing them with new fix it?
Should i check something else as well ?

TheCyberQuake, 2025-05-29:
> Checked with multimeter and it looks like there is short on both ceramic capacitors(sp1/2).
> Will replacing them with new fix it?
> Should i check something else as well ?

This isn't tech support for damaged hardware. This isn't a forum, its a bug report. Search for help online as it's no longer related to the project here

treyus30, 2025-11-15:
> Nope. Had several of us have already found that error code just means CLK is shorted to ground. Your replace fix was likely just the fact that you undid the old one with a bad solder and redid the [...]

I'm getting this error but the console boots to the normal screen. Wouldn't a short to ground prevent boot completely?

aguerrero31, 2026-10-30:
Quick edit: To be clear, this was a device that was not working prior and the modchip was installed to attempt to help diagnose the issue. This could also apply to a device that was previously modchipped and has since stopped booting, though.

If anyone is attempting to repair a dead switch, this error code can also mean that your NAND is dead. I just helped someone out with an OLED where they had short long long and 136mA of power draw on boot. Everything on their modchip install checked out, so that pointed to a dead eMMC, dead APU, or a broken connection somewhere between the eMMC and APU. They removed the eMMC and swapped it with one from another device, and the console now functions (albeit as an offline-only console due to not having an original eMMC backup).

If you're seeing similar symptoms and have the skillset and tools, remove the eMMC and check all of the connected pads with diode mode. If you have 0 on any data lines then you'll know the APU is done for. If you have 0L on any data lines then you'll need to pull the APU and verify a connection between the eMMC and APU pads on the board. If everything checks out then replace the eMMC, and use Sthetix's Level 3 restore if it's a new (or "new") eMMC to format and populate or repopulate the memory's storage. As noted above you'll have an offline-only console at this point, but it's better than a dead one 😄