Ansem-SoD/Picofly #32: error code *=*(short long short)

fesily, 2023-10-29:
Switch lite get the error code. 
I don't have any idea about this problem.
Data0: 0.480
3.3v: 0.398
CMD: 0.515
CLK: 0.530
RST: 0.440
This is what I measured on the 2040 motherboard

fesily, 2023-10-29:
I've replaced 3 motherboards and it works perfectly fine on other switch lites

Quas7, 2023-10-29:
EMMC init fails.
What do you get on B (rst)? 
Try to clean all flux residue again maybe.

radius611, 2023-11-11:
I have this error code as well on V1 switch. I have tried making sure my board is as clean as possible. I've checked the CPU solder joints as well.  Switch boots fine if eMMC is directly connect, but won't boot with picofly. I've tried 2.73 and 2.7.5.

hanswurst31, 2023-11-19:
i also have this problem on patched erista. everything looks smooth. tried another picofly. same error. wonder if the flex cable is faulty?

closed 2023-12-05 by fesily