Ansem-SoD/Picofly #33: Error code ==* (Support)

Harry040502, 2023-11-01:
I’m getting this issue (glitch no worky Check mosfet)

It then boots into the default dance OS. Was wondering if it’s my crappy soldering (probably this) or something else. I can provide photos of the swag soldering if you think it’s that.

Appreciate any support (other than ppl just saying my soldering is bad without explaining the actual faults (not productive))

not an issue with this software issue with me.

Harry040502, 2023-11-03:
[image]

Quas7, 2023-11-03:
Left outside capacitor seems not to be soldered properly. Do you have continuity checked all solder points?

And flux residues should be cleaned off.

Harry040502, 2023-11-03:
[image0.jpeg]
Same issue after trying to fix.
- Harry

[https://www.plymouth.ac.uk/images/email_footer.gif]<http://www.plymouth.ac.uk/worldclass>

This email and any files with it are confidential and intended solely for the use of the recipient to whom it is addressed. If you are not the intended recipient then copying, distribution or other use of the information contained is strictly prohibited and you should not rely on it. If you have received this email in error please let the sender know immediately and delete it from your system(s). Internet emails are not necessarily secure. While we take every care, University of Plymouth accepts no responsibility for viruses and it is your responsibility to scan emails and their attachments. University of Plymouth does not accept responsibility for any changes made after it was sent. Nothing in this email or its attachments constitutes an order for goods or services unless accompanied by an official order form.

Harry040502, 2023-11-03:
[image]

Quas7, 2023-11-04:
Maybe check youtube for "diode readings" for nintendo switch and measure these first?

Harry040502, 2023-11-05:
Turns out the flex cable connector was dodgy 😢

[https://www.plymouth.ac.uk/images/email_footer.gif]<http://www.plymouth.ac.uk/worldclass>