Ansem-SoD/Picofly #34: picofly no sd OFW Boot broken on 17.0.0

1stdevfriend, 2023-11-08:
Picofly flashes a blue light then purple

OFW automatically updated to 17.0.0 now picofly doesn't boot

Uploading VID20231108191914.mp4…

Quas7, 2023-11-08:
did you update your SD card as well?
I think, you need to setup hekate (https://github.com/CTCaer/hekate/releases) to fit to 17.0.0 as well to boot

1stdevfriend, 2023-11-08:
I don't think that's the issue cause picofly should show no sd card on removal @Quas7 

Instead, it's booting directly into OFW

Quas7, 2023-11-08:
Ah ok. Googled a bit and 17.0.0 is actually not working at all with picofly.
Guess Nintendo did a good job locking with 17.0.0.

1stdevfriend, 2023-11-08:
Hmm not sure cause i read some articles which said picofly can be updated @Quas7

Quas7, 2023-11-08:
Yes, updated firmware for the picofly is very likely needed.

Quas7, 2023-11-08:
looks like there are ppl with success even without updating and some have the issue above.
https://gbatemp.net/threads/picofly-a-hwfly-switch-modchip.622701/page-668#post-10265952

Maybe it is worth a try to flash the current 2.75 pico-firmware again in order to let it start re-traing the glitch timing?
@1stdevfriend Do you actually run already on 2.75?

Quas7, 2023-11-08:
Just to state the obvious... the official picofly firmware is located here: https://github.com/rehius/usk/releases/tag/PicoFly_2.75_2023.09.09_08-16

hongtm77, 2023-11-20:
> Just to state the obvious... the official picofly firmware is located here: https://github.com/rehius/usk/releases/tag/PicoFly_2.75_2023.09.09_08-16

Hi, i got the exact same issue where my firmware was 2.74. And sometimes will boot to ofw. I tried to upgrade to 2.75 using the link provided and still the same. Will it be the picofly chip issue or all are the same? I saw some post suggesting that we can wire the sda and scl i2c. Im not quite sure and cant find any diagram for it 🤔🤔

closed 2024-02-07 by 1stdevfriend