Ansem-SoD/Picofly #42: Switch OLED witch eMMC error *=* (Change the mmc?)

lucasromeiro, 2024-02-03:
Hello, I have technical assistance and I do switch unlocking.
But, 2 consoles arrived here from people who tried to do the process at home and damaged the console.
None of them turn on and PICOFLY gives the error (*=* No eMMC block 0 read (eMMC init failure?)).
I imagine the eMMC was permanently damaged.
Has anyone managed to recover from this type of problem??

I had an idea yesterday to buy a new EMMC and replace it, but I don't know if that would work. Has anyone already done this?
I'll leave here a link to one I found on Aliexpress:

[Link 1](https://pt.aliexpress.com/item/32901707646.html?aff_fcid=5e707c44aab94157b644c9df4379d4b6-1706971991301-00300-_oEHZXTk&aff_fsk=_oEHZXTk&aff_platform=api-new-link-generate&sk=_oEHZXTk&aff_trace_key=5e707c44aab94157b644c9df4379d4b6-1706971991301-00300-_oEHZXTk&terminal_id=c33d85b70ffa486894ac5a90cfc2610a&afSmartRedirect=y)
[Link 2](https://pt.aliexpress.com/item/1005005735298542.html?spm=a2g0o.productlist.main.37.4f0c505dpo83aE&algo_pvid=807f237d-ff0a-4462-9542-d2e2eddbf54f&algo_exp_id=807f237d-ff0a-4462-9542-d2e2eddbf54f-18&pdp_npi=4%40dis%21BRL%2135.36%2135.36%21%21%216.85%216.85%21%402103081017069719337997363ef6e9%2112000034158533657%21sea%21BR%21141067012%21&curPageLogUid=rCz0veva0vBs&utparam-url=scene%3Asearch%7Cquery_from%3A)
[Link 3](https://pt.aliexpress.com/item/1005005812975528.html?spm=a2g0o.productlist.main.27.4f0c505dpo83aE&algo_pvid=807f237d-ff0a-4462-9542-d2e2eddbf54f&algo_exp_id=807f237d-ff0a-4462-9542-d2e2eddbf54f-13&pdp_npi=4%40dis%21BRL%21443.34%21376.85%21%21%2185.88%2173.00%21%402103081017069719337997363ef6e9%2112000034441977648%21sea%21BR%21141067012%21&curPageLogUid=A5q9NGuqG7j5&utparam-url=scene%3Asearch%7Cquery_from%3A)

I would like to do this, but I don't know if it works.
Or if there is a way to recover the damaged one.

impeeza, 2024-02-03:
Yes could work, but you will need a backup of NAND or at least the console keys and follow the procedures to recreate a nand using a donor one. There are several on gbatemp

lucasromeiro, 2024-02-03:
> Yes could work, but you will need a backup of NAND or at least the console keys and follow the procedures to recreate a nand using a donor one. There are several on gbatemp

@impeeza 

But, how can I back up the memory and get the keys if the console won't turn on? Could I use key and backup from another console?
What is donor? I've never done anything like this, I don't know what tools are needed, I thought it would just be changing the nand.

Roigar, 2024-02-23:
> I'm having the same problem. I believe it's bad soldering or cables shorting, gonna get it fixed this weekend. Hopefully it's not a dead eMMC.
> 
> My switch doesn't even boot into the OFW though, it's permanently stuck on black screen with the _=_ No eMMC block 0 read (eMMC init failure?) error. I even unplugged the modchip cables and tried booting the console, but no luck. It was working fine a couple of days ago.
> 
> I will try unplugging the battery for 2-3 hours and then try booting the console with the modchip unplugged.

So I managed to get my Switch working again after letting its battery fully discharge then charging it back up again to full.

I hope I don't have the same issue again, then I'll have to take it to the repair guy again.

cmorenot, 2024-08-31:
Tengo este error en mi nintendo Switch Oled,prende en azul y luego en amarillo dos parpadeos cortos y uno largo (**=) alguna solución a este error??

Jhoan0822, 2025-10-26:
> Tengo este error en mi nintendo Switch Oled,prende en azul y luego en amarillo dos parpadeos cortos y uno largo (**=) alguna solución a este error??

Hola, lo pudiste arreglar?