Ansem-SoD/Picofly #44: What is wrong with 2.7.4?

caden3ds, 2024-03-04:
My modchip came with 2.7.4, do I need to downgrade it to 2.7.3?

caden3ds, 2024-03-04:
Idk how to label this as a question and not an issue sorry

mxarment, 2024-03-30:
I have the same question. Ive created another issue https://github.com/Ansem-SoD/Picofly/issues/45 and Im now thinking it can me related with this you are talking about, Idk if I should go to 2.7.3

jablochniyvorishka, 2024-05-20:
[v2.74 + 300 mhz precision]

 rp2040 may be not stable at 300mhz
 
 use 2.7.3 or 2.7.5

jablochniyvorishka, 2024-05-21:
> > [v2.74 + 300 mhz precision]
> > rp2040 may be not stable at 300mhz
> > use 2.7.3 or 2.7.5
> 
> Where do i find the newer firmwares?

https://github.com/rehius/usk/releases/tag/PicoFly_2.75_2023.09.09_08-16

Ixolaro, 2024-06-17:
are they safe to install with toolbox?