Ansem-SoD/Picofly #49: Picofly OLED install short short long error

Samybigio, 2024-07-21:
the issue is that sometimes the switch will boot in ofw sometimes with the following error code (**=) which is not even listed in the readme. i have an OLED switch and more info is in the video

https://github.com/user-attachments/assets/a1335fcc-97ee-42a7-9d71-00214da8f12f

Samybigio, 2024-07-21:
update: i did some research and apparently its an issue related to picofly modchips, seems like it could be fixed with a firmware update

CamarilloJsr16, 2024-07-22:
I have the same problem, either in sysnand or emunand 18.0.1, when I reboot the switch it starts directly in ofw and I can't start in hekate by any means or combination of buttons. Were you able to solve it? It happens to me on the oled switch

CodeHub1004, 2024-07-29:
I am also having this problem. I have installed many OLEDs and first time this is happening. 1 out of 15 boots it goes into OFW. When it boots to hekate everything works as it should.

Samybigio, 2024-08-29:
> I have the same problem, either in sysnand or emunand 18.0.1, when I reboot the switch it starts directly in ofw and I can't start in hekate by any means or combination of buttons. Were you able to solve it? It happens to me on the oled switch

sadly not, i am just living with this error popping up every once in a while

CamarilloJsr16, 2024-08-29:
> sadly not, i am just living with this error popping up every once in a while

I solved it by updating the fw to 2.75, I found it on google.