Ansem-SoD/Picofly #60: =*** eMMC init failure

silentcovenant, 2025-01-24:
I installed a waveshare RP2040-Tiny and I'm getting the =*** eMMC init failure. What do I do?
I previously had the v6 mod like sthetix installed and it was working great until it started not booting into hekate after leaving it docked for a while.

What do I do about this error? I didn't know where else to ask.

marcbenedi, 2025-02-02:
I also wonder something similar, is it possible to boot into hetake with a faulty/non-working eMMC?

JaschB6777, 2025-03-02:
I have the same error, freshly installed, tried 2 PicoFly V3 chips, measured everything with a multimeter, but still get =*** and it starts in OFW.

Help Please

marcbenedi, 2025-03-02:
> I have the same error, freshly installed, tried 2 PicoFly V3 chips, measured everything with a multimeter, but still get =*** and it starts in OFW.
> 
> Help Please

Hi,

I ended up getting it working but I am not sure what exactly happened.
I resoldered SP1 and SP2 (for a V1 switch) around 5 times (remove -> clean everything -> solder again etc) until it worked. 

One thing to take into account is, make sure your eMMC is not shorting with the picofly. After it worked, sometimes it would freeze mid game etc, until I put tape between the chips.

JaschB6777, 2025-03-02:
I have also resoldered SP1 and SP2 3 or 4 times, but okay, at least now I know what to focus on and will try a few more times. Thanks!

JaschB6777, 2025-03-03:
I managed to fix it. It turned out that I pressed the DAT0 flex down too hard while soldering, which connected multiple pins of the eMMC. The trick is to place the DAT0 flex gently