Ansem-SoD/Picofly #61: Short long long error, completely RANDOM

T0T0W, 2025-01-25:
Please someone help me before i kill myself.

I get the error Short long long.

The thing is, it is completely random, sometimes it works, sometimes it doesn't (it will work perfectly and then completely stop working for a while and then work again.)

Even stranger, when it works perfectly, everytimes i connect the screen it will suddently stop working, it won't glitch at all.

My soldering is fine i think. In the two videos keep in mind i haven't touched the motherboard at all between each try. I just unplugged and plugged back the charger.

I've already removed and re-sit the dat0 adapter. Same error.

Here's the values on the picofly chip:

3v3: 0.946v
D: 0.841v
A: 0.842v
C: 0.844v
B: 1.615v
Ground: 0.001v

[image]
[image]
[image]
[image]
[image]

https://github.com/user-attachments/assets/2565b9e6-9e28-4422-aad6-bbd24bff676f

https://github.com/user-attachments/assets/0e622cd0-873a-4ec1-8ddb-1cda42932c6a

T0T0W, 2025-01-25:
I managed to go to the "No SD Card" screen. Without changing anything.

Then after a restart it doesn't work anymore. Error Short Long Long. Cool.

MommoIsItaliano, 2025-01-25:
Hello, I happen to be new here, so bear with me.
I happen to be experiencing the same problem except I didn't do the mod myself.
A couple of things:
- that code is `*== No eMMC CMD1 request (poor wiring, or dead CPU)`. 
- it seems from this thread here: https://github.com/Ansem-SoD/Picofly/issues/31 that the problem is the CLK line is touching something that goes to ground. I recommend you read said thread in detail as it may help you more thank I could. It seems that it could be the shield around the SoC shorting the CLK line, or alternatively you may have exposed too much of the CLK trace, thus also exposing the ground plane around it, thus immediately grounding the contact.

I don't know if this helps at all but if you do manage to figure out a solution, would you mind telling me?

T0T0W, 2025-01-25:
> * o

hey, i've booted the console without both heatshields (emmc and cpu) while making sure it doesn't short anywhere and it still did the same. I've also checked if any of my points are shorted to ground and they aren't, also my CLK line is perfectly scrapped. I don't ahve any pics but i did it with a needle and a microscope, and i was VERY gentle, it looked perfect. HOWEVER, the issue seems to have been fixed after a good cleanup of everything. I still had some flux on the Dat0 and the other flex cable (clk, 3.3v etc)

I managed to reassemble the console and it seems to glitch fine everytimes...

It's ridiculous tbh. I have no idea if that was the issue or not. I doubt flux can give boot problems but whatever. I just hope it's fixed for good.

MommoIsItaliano, 2025-01-25:
I'm very glad to hear that! Enjoy it. 
I'm going to give mine a check with a modder near me in order to figure out what this is and put a nail in this coffin.

Richie-lem, 2025-04-18:
Exactly same issue here, Oled picofly, everything works fine but at some boot *== happens and I dont know why at all =(